sharkduino/Arduino_Animal_Tag #3: Parser usablity

WLaney, 2016-07-21:
Lets make the parser easer to use. Don't worry about fancy options or anything, just simple and stream line. Additionally maybe make the output files dated

TheBen27, 2016-10-11:
Awhile back, I made it print the readme upon incorrect invocation awhile ago, so it's streamlined.
The output files are not dated, but they probably should be... I'm renaming this issue to reflect what's left to do.

TheBen27, 2017-02-13:
My final format will go something like this:

If the user does not specify the output filenames, the names will follow the format `[tag name]-[date]-[data/header].csv`

WLaney, 2017-02-13:
I think `[date]-[tag name]-[data/header].csv` would be slightly better. That way computer file systems will but all our data in chronological order. It's really up to you though, and it might be better to have the file system sort the same tags together. 
Also I know I'm kind of splitting hairs here.

TheBen27, 2017-02-13:
No, that makes more sense. Thanks

TheBen27, 2017-02-13:
Done with commit [d925b5](https://github.com/WLaney/Arduino_Animal_Tag/commit/d925b5db13afa2866aebc4f1dfb21be0055a7c78)

closed 2017-02-13 by TheBen27