sharkduino/Arduino_Animal_Tag #4: Delay start

WLaney, 2016-07-21:
use pin two on the RTC to set a time for the tag to turn on

TheBen27, 2017-02-17:
Putting this in the configuration sketch, see the issue on gyro rate changes for more

TheBen27, 2017-04-12:
In configuration sketch, integrated into main sketch.
**Reconfigure your tag** so your tag doesn't sleep for 2^32 seconds on startup.

closed 2017-04-12 by TheBen27

WLaney, 2017-04-12:
To confirm, we need to run the configuration sketch on all existing Sharkduinos that use this version? If I do re-run the configuration will it be backwards compatible, or will we need to reconfigure again if we switch the code used on board back to an older version?

TheBen27, 2017-04-12:
My fix only applies to Sharkduino 3.0, so... um. Yeah, I'll go hack the
change into 2.0 and 1.0 tomorrow.

WLaney, 2017-04-12:
We can say that turn on is a feature that can only be used in code 3.0 forward, and thus only works with hardware V2.1 forward if you want. I was just curious