sharkduino/Arduino_Animal_Tag #6: easy frequency change

WLaney, 2016-07-21:
Make it so the frequency of sampling is easy to adjust. Don't worry about under sampling, just throw away the extra data. It's fine

TheBen27, 2017-02-17:
Decided to add the frequency-changing feature to the configuration sketch.

If the tag is eventually sealed off, we can't use a configuration file, so there's no reason to invest too much time in that. Later, we can put the configuration sketch into the main sketch and facilitate communication through Bluetooth (or USART) while it's charging

TheBen27, 2017-02-25:
Tested configurable sample rate. The sample rate is about 7ms off consistently, but it was in the original sketch. I added three sample rates: 12.5, 25, and 50Hz. All 3 seem to work, so this issue is closed.

closed 2017-02-25 by TheBen27