sharkduino/Arduino_Animal_Tag #8: If tag resets, it appends new set of reads to old

TheBen27, 2016-07-21:
If the tag is ever reset during operation, it will append the new set of reads to the old file instead of creating a new one. This issue was negligible back when we could manually go in and separate the two sets of reads into two files - but the size and binary nature of the files will make this difficult, especially for the end users.

WLaney, 2016-07-21:
I think we should make sure that it puts information into the file letting us know the reset happened

TheBen27, 2016-07-22:
In a sense, that already happens automatically - the header is written on each reset. We can have the parser split the input into multiple files and warn the user that it happened.

The alternative is to have the Arduino itself create new files upon resetting. Our philosophy so far has been to keep the raw data opaque and have the user use the parser, so it's easier for us to use the parser.

TheBen27, 2017-02-13:
Arduino now creates new files upon resetting. I know what I said, but the Arduino code's easier to change than the horrid parser code. For now, consider the issue closed.

closed 2017-02-13 by TheBen27