sharkduino/Arduino_Animal_Tag #10: recored if tag stops recoding due to no sd or dead battery

WLaney, 2016-08-02:
I don't know if this is possible and I don't any good ideas on how to do this. Keep in mind that both of these could happen, and we want to know which one happened first. Is is possible to write that the sd is not available to system memory in a way we can easily access once power is restored to the tag?

TheBen27, 2016-08-02:
The "System memory" you're thinking of is EEPROM, and it's perfectly suited
to this task. I don't know how to detect when we're almost out of power,
but the pseudocode for the SD stuff is something like this:

On update:
    Do short-term reads until buffer full
    If SD card accessible:
        Long-term SD writes
    else:
        Save "we can't access the SD" to EEPROM
        If the RTC is accessible, save the current time in EEPROM

We can then write diagnostic code (either a separate sketch or the main
code) that reports these errors to the user. I'll get on that at some point
in the future...

Is there a hardware method for determining how much battery we have left?

> I don't know if this is possible and I don't any good ideas on how to do
this. Keep in mind that both of these could happen, and we want to know
which one happened first. Is is possible to write that the sd is not
available to system memory in a way we can easily access once power is
restored to the tag?
<https://github.com/WLaney/Arduino_Animal_Tag/issues/10>, or mute the thread
<https://github.com/

WLaney, 2016-08-02:
For the battery level we could try to do something with measuring the battery voltage relative to the system voltage and determining the battery percentage from that. The problem is we can't do that with the battery just connected to GND and RAW, we need to connect to an analogue pin. I can do some tests when I am back at WM and make sure it would work though. 

Alternatively there have only been two ways that the system has failed that give us data, but not as much as expected. We could just recored the SD status and make the assumption that if it's not the SD it's the battery. I think that's fairly safe.

TheBen27, 2016-08-02:
Mm.  And we could get the time from the file in the case of power loss...
makes sense.