sharkduino/Arduino_Animal_Tag #15: [Dumb Idea] Accelerometer FIFO + Gyro FIFO = Sleepier Arduino

TheBen27, 2016-12-27:
As it stands, the gyroscope has a FIFO buffer and the accelerometer doesn't. This means that the Arduino has to wake up 25 times a second to store the accelerometer's data, which partly negates the benefits of having a FIFO gyro in the first place.

A couple of different accelerometer models (like the MMA8451) have a 32-sample FIFO buffer, just as the FXAS gyro does. If we use both FIFO buffers, the Arduino will only have to wake up about once every 1.25 seconds at 25Hz.

There are a few concerns here:
 * Do accelerometer models with a FIFO buffer use more power?
 * How much power does using FIFO really save? It's theoretically more efficient (fewer wake-ups, fewer I2C requests), but that doesn't mean anything...

The first concern can be checked via datasheet and empirical tests. I tried to check the second one awhile back using the old gyroscope, and the results were inconclusive. I should try again with the new one and learn how to actually use an oscilloscope.

The Arduino can still be used as a buffer in this setup.

I'll get the gyro done first before looking into this further. Just spouting out dumb ideas for now.
--Ben

WLaney, 2016-12-27:
Looking at the data sheets the MMA8451 and the MMA8452 seem nearly identical. They appear to have the exact same power rating, and use the same circuitry. The only differences I found on first glance are that the MMA8451 has higher resolution, higher sensitivity, an FIFO buffer, and is $0.30 more expensive. If you want we can pick up some MMA8451s and just pop them onto the existing boards with no more work on my end. 

The only reason to hesitate is that I haven't really looked at accelerometers to closely yet, and may decide that I want to change to a totally different one. I don't want you to do a lot of work developing for a chip that we're not going to be using for very long.

TheBen27, 2016-12-27:
I get that. We can do some more research and figure out what's reasonable.
Until then, I'll keep working on the gyro.

WLaney, 2017-02-10:
I have ordered some MMA8451 chips. With any luck I will have some working ones on boards in the coming weeks for you to experiment with.

TheBen27, 2017-02-10:
Nice. I looked at the data sheet once. It's not a drop-in replacement, but
I will figure it out.

WLaney, 2017-02-22:
A Sharkduino V2.1 with a MMA8451 is assembled, and seems to be working

TheBen27, 2017-03-30:
FIFO buffers have been integrated into the code, although the interrupts aren't. For this issue's intents and purposes, this is perfectly fine.

closed 2017-03-30 by TheBen27