sharkduino/Arduino_Animal_Tag #18: initial MMA8451 tests

WLaney, 2017-02-17:
Hey the MMA8451 accels came in today. Do we have a way to test that they are working? I can run the I2C scanner and make sure it shows up, but do we have any more sophisticated tests we want to run? 

On a house keeping note, this is not part of the [Accelerometer FIFO + Gyro FIFO = Sleepier Arduino](https://github.com/WLaney/Arduino_Animal_Tag/issues/15) issue because it only deals with the MMA8451, not integrating the MMA8451 in such a way to make the Arduino sleep more.

TheBen27, 2017-02-17:
There's an existing library for this at:

https://github.com/akafugu/MMA845XQ/tree/master/MMA845XQ

Since its functionality is not congruent with our needs, I'll probably
write my own. After the FXAS gyro, it shouldn't be too bad.

WLaney, 2017-02-17:
Since that is a generalized library for MMA845 family of accelerometers, does that mean that the old code should work with the new accel? Or at the very least reading data works the same way?

TheBen27, 2017-02-22:
I believe the old library would work, at least for reading data. It might break down for more esoteric features, but we could use it for testing

WLaney, 2017-02-22:
The old library did not work as is. I don't know if it is a software or hardware problem, but the accel does show up on the I2C scanner.

TheBen27, 2017-02-22:
What about the new library?

WLaney, 2017-02-22:
I didn't get a chance to test it out. I'll do that when I have time and get back to you.

WLaney, 2017-02-27:
I got this to work with the library you linked to. I switched the I2C address to 0x1D and had to edit the .h file to change the path to the Wire library.

WLaney, 2017-03-28:
Since we are moving forward with the MMA8451 would it be possible to get this test folded into the test_all sketch in a future release? Since I am going to be assembling a lot of Sharkduino V2.2s, which have this sensor, I would like a more convenient way to test it, if possible.

TheBen27, 2017-04-04:
Folded into the test_all sketch, pushed to GitHub.

closed 2017-04-04 by TheBen27