sharkduino/Arduino_Animal_Tag #21: Gyro data not correctly recorded: Aug 2016-Feb 23, 2017 [Master]

TheBen27, 2017-02-24:
This bug has already been solved, but it's impactful enough to warrant an issue. Between about August 2016 and February 23, 2017, tags using code from the master branch did not correctly record gyroscope data.

This is because main sketch sampled at 12.5Hz. Although both the accelerometer and the gyroscope were sampling at 25Hz, the gyroscope's buffer filled up too quickly and captured only the first half of the data. The parser then stretched that half to fill in the gap.

Again, it doesn't impact the 25Hz branch or the shark data collected in July, only the Master branch. The issue is now fixed, but if you made any test sketches during that time period and noticed some odd behavior, you now know why.

WLaney, 2017-02-24:
Just to be clear, only the gyro data in that time frame was affected? The accel did not use a FIFO buffer, so it should be fine.

TheBen27, 2017-02-24:
That's correct.

Centigonal, 2017-03-13:
Hey! I've been trying to record some sample data with the Sharkduino V2.0a, and it seems like gyro data is still missing in ~25 sample increments. I tried commits 394214e, 965798d, and 820ea9e, and experienced the same issue with each. Maybe we could troubleshoot this later!

TheBen27, 2017-03-13:
Does the data follow a pattern of [32 samples, 32 blanks]? If so, it's working as intended. I added gyro turnoff code and, to test it, had the gyroscope turn off for every other set of reads. That's probably not what we want, so I'm going to remove it in my next commit.

(32 might not be the exact number, I think it's more like 48)

TheBen27, 2017-03-13:
Commented out turn-off code. Check it on your end.

Centigonal, 2017-03-13:
Works great! Thanks so much! :D

closed 2017-03-30 by TheBen27