sharkduino/Arduino_Animal_Tag #22: Code with no SD writes

WLaney, 2017-03-13:
Would it be hard to hack together some code that only reads data, but never writes to the SD card? I want to use it for some power testing

TheBen27, 2017-03-13:
That's doable. To clarify: you want a sketch for the 1.1 that reads data,
then throws it away instead of writing it?

I'll do it tonight.

WLaney, 2017-03-13:
I want a sketch for V2.0, you can do V2.1 as well if you want. I doubt that it matters very much for what I'm looking at though.

TheBen27, 2017-03-13:
Just made the no_sd branch and took out all the SD code.
On a related note, did you know that like half our sketch's space comes from the SD library?
I mean, not like space is a real priority but maybe flash memory _is_ the way to go
Anyway, check it out and close it if you find no problems

closed 2017-03-14 by WLaney