sharkduino/Arduino_Animal_Tag #26: Tag resets on SD card insert

TheBen27, 2017-03-25:
Hey, guys.

As you know, I don't really know electronics, so "power fluctuation" is a bit of an embellishment on my part. Regardless, when I insert an SD card into my Sharkduino, the power light flickers. It also seems to disconnect and reconnect the device - I sometimes have to switch the port in the Arduino IDE (between /dev/ttyUSB0 and /dev/ttyUSB1).

I don't know for how long this has been occurring, but I just noticed it earlier today.

Is this something we should worry about? Could this be related to our old problem with the device resetting while recording data?

Thanks.
--Ben

WLaney, 2017-03-25:
Isn't your sharkduino the one where the gyro doesn't work? If so it's already broken and I don't know what the under lying cause of the problem is. The issue could be manifesting in diffrent ways. You should stop using it, and grab V2.1b from small.

TheBen27, 2017-03-25:
Fair enough. I'll pick it up tomorrow.
For now, I guess we can consider this case closed.

closed 2017-03-25 by TheBen27