sharkduino/Arduino_Animal_Tag #27: Versioning System

TheBen27, 2017-04-04:
## Sharkduino Software Versioning Scheme

> Each hardware version gets its own branch and set of releases. Bugs are patched retroactively; new functionality isn't. Versioning is `hardware.functionality.patch`.

As we discussed, my initial versioning system, which tried to follow the hardware's versioning as closely as possible, was inadequate for two reasons:

* The development of the software is not congruent with the development of the hardware.
* We can't easily patch code retroactively.

This is the new system we agreed on as I understand it. Add your questions or comments below - the new system will take effect on Wednesday or Thursday.

### Release Rules

Releases will occur when:
* I finish moving the code to a new hardware configuration.
* Major functionality has been added.
* A serious bug has been patched.

Each release will be versioned as `hardware.functionality.patch`. A `hardware` release will correspond to a new hardware configuration, but the number will *not* correspond to the version of the new Sharkduino tag.

Software Version | Supported Tags
---|---
 <1.0 | Prototype tags are not supported
 1.0 | All tags between `[1.0, 2.0)`
 2.0 | 2.0, 2.1
3.0 | 2.2

`functionality` releases are for major functionality additions, like buffering or interrupt support. `patch` releases are for bugfixes that compromise data or battery life.

### Release Branches

New hardware releases will be given their own "release branch". The branch will solely contain all releases for that hardware as tagged commits.

Bugs will be patched retroactively on each supported hardware configuration through a new tagged commit to that branch. Barring exceptional circumstances, new functionality will only be added to the latest hardware configuration.

WLaney, 2017-04-08:
Sounds good. Are you clear on what hardware versions have what sensors and functionality?

TheBen27, 2017-04-08:
I think so. The INT pins might be wrong (see 2.1 and 2.2)

Arduino 1.0

   - MMA8452Q Accelerometer
   - L3GD20H Gyroscope
   - DS3234 RTC
   - TMP102 Temperature Sensor
   - MS5803 Temp/Pressure Sensor
   - Push-toggle (?) microSD slot

Arduino 2.0

   - MMA8452Q Accelerometer
   - FXAS21002 Gyroscope
   - DS1339B RTC
   - External MS5803 Temp/Pressure Sensor
   - Push-pull microSD slot

Arduino 2.1

   - New accelerometer: MMA8451Q
   - MMA8451Q INT1 pin wired to Arduino pin 3

Arduino 2.2

   - FXAS21002 INT1 pin wired to Arduino pin 2

WLaney, 2017-04-08:
Yeah just a couple things:
V2.2 FXAS21002 INT2 pin is wired to Arduino pin 2, not FXAS21002 INT1 
V2.0 the MMA8452Q INT1 pin is wired to Arduino pin 3
RTC int is connected to Arduino pin 2 for all RTCs and versions
The technical name of the uSD connector we used in V1.0 is push-push

All this stuff is in the documentation on the hardware repo, but if you want to update your list above, and add it as a quick reference that would be great. It is a little overwhelming to go through the documentation for everything at this point.