sharkduino/Arduino_Animal_Tag #28: Look into lowering clock frequency

TheBen27, 2017-04-07:
By changing the fuses on the Arduino, we can choose to use very low-frequency clocks (128Khz, even) without changing the hardware. The Arduino spends most of its time in I2c and SPI transactions, which could be the cause of some wasted cycles (waiting for the transaction to complete).

* Can we lower the clock frequency without impacting data collection, especially for long periods of time?
* If so, does it reduce power consumption significantly?
* Programming the fuses will require us to go "off the reservation" by using a makefile instead of the Arduino IDE. Even if it works, is it worth the extra complexity?

WLaney, 2017-04-08:
I think this is more of a discussion topic. You should email this to the list or we can talk about it at the next meeting. 

I would do so research into what the power savings actually are. Also once we are out of the Arduino IDE we should probably think about changing microcontrollers.

TheBen27, 2017-04-08:
Actually, we might not have to move out of the IDE? AVR LibC has some
headers that let you set the fuse bits within the source code.

The problem, of course, is that Arduino probably sets those fuse bits in
its own code. So it's anybody's guess as to what will happen.

Like I said, I'll get into it more over the weekend.

WLaney, 2017-11-07:
Me and Kelvin have been looking at the 328p clock speed in [Tribesat](https://github.com/tribesat). If you're still interested in this you should maybe talk to him.