sharkduino/Arduino_Animal_Tag #35: Docs for config sketch

Centigonal, 2017-04-15:
Is there documentation for the configure sketch? Like what all the different options do, which ones work/are tested, etc.? If not, can we make a readme or something?

Thanks!

TheBen27, 2017-04-15:
There is currently no such document. Since you're asking about this, I have
to assume that you are either confused about some aspects of the sketch or
are worried that others might be, which I consider a product of my
(admittedly slapdash) UI.

I'd rather not require my users to consult a manual to use my configuration
sketch, so I'll probably spend some time updating it for usability in the
coming weeks.  ... Not this week, though. I've put in my ten hours, thanks
much.

Centigonal, 2017-04-18:
No worries! My concern is that certain parameters (like bias) might not work, and others (like HQ accelerometer) might need some explanation for people new tot he project. It's definitely not a high-priority item, but it would be ideal if someone like Schenck or a marine scientist could run the config sketch without coming to us with questions.

TheBen27, 2017-11-04:
Individual configuration items now have short (2-3 sentence) explanations. Not sure if this is really enough, so I'd like you (or someone else) to take a look at it and see if you still have questions that the UI fails to answer.

Thanks for your help in this.