sharkduino/Arduino_Animal_Tag #39: I2C address list

WLaney, 2017-09-13:
We need to confirm the address in [this document](https://github.com/sharkduino/Arduino_Animal_Tag/blob/master/papers/hexdigits.txt), and add the RTC address to it.

WLaney, 2017-10-21:
@TheBen27 has removed the document referenced in the original issue. It did not have correct information, but should be replaced. It can also be moved into a hardware repo if that makes more sense.

TheBen27, 2017-10-21:
Just pushed `papers/I2C_Reference.md`. However, it is incomplete, as the addresses depend on the values that certain pins are set to. I think these pins are set in hardware, not software, so I think we should move this document to the hardware repo and replace the version in this repo with a link.

Since the values of these pins will change with hardware versions, it makes sense to place this somewhere where hardware people can easily access and modify it.

WLaney, 2017-10-22:
The pins are set on the hardware level, and have stayed constant through all revisions of the tag. I think the reason the address list is in this repo is because we have had bugs in the past from using the wrong address in the code. 

That being said I agree it would make more sense for this to be in [sharkdunio_layered_desgines](https://github.com/sharkduino/sharkdunio_layered_design). We can close the issue here and reopen it over there.

closed 2017-11-01 by TheBen27